oshwabadge2020/Badge-PCB #3: use case for microphone?

pdp7, 2019-12-13:
I see the SPH0645LM4H-B is a microphone.

What sort of functionality where you thinking about for that?

@mwelling had mentioned TensorFlow Lite when we were talking about Artemis.  I wonder if that is still possible on the nRF52840

mwelling, 2019-12-13:
https://www.tensorflow.org/lite/microcontrollers

It should be possible with a fair amount of work. 

I tried the Tensorflow Lite demo on the Edge Badge from Adafruit and it could not recognize me saying "yes".

I think that there were similar complaints about the SparkX Edge demo.

If we want to teach the module different words we will have to create a new model and quantize it for use with Tensorflow Lite.

pdp7, 2019-12-16:
@acamilo @mwelling OK, let's keep this on the "nice to have list".  If cost or board layout becomes issue, then I think we could drop the microphone.

sound ok?

mwelling, 2019-12-16:
Yes.

closed 2020-01-13 by pdp7