oshwabadge2020/Badge-PCB #5: layout: 2 boards or 1 board?

pdp7, 2019-12-23:
I see that @mwelling is experimenting with a two board design based on the initial design by @acamilo.

Alex / Michael - do you think this will increase the complexity for SparkFun to assemble?  could we get away with one board, or do we have too many features to pack in to that space?

@nseidle @npoole would two boards for the OHS wristwatch badge be acceptable for sparkfun assembly? 

thanks!

mwelling, 2019-12-23:
The boards will be a single assembly connected by mouse bites. It should not be any more difficult to assemble than a single board.

pdp7, 2019-12-24:
fyi - from the project chat:
[Stacking boards and wrist mounting options](
https://hackaday.io/project/168483-open-hardware-summit-2020-badge/log/172231-stacking-boards-and-wrist-mounting-options)

closed 2020-01-13 by pdp7