oshwabadge2020/Badge-PCB #7: Generate gerbers for review

nseidle, 2019-12-23:
I am not a proficient Kicad user. If possible, could you generate gerbers so I can review?

mwelling, 2019-12-23:
Yes we can provide Gerbers and PDF plots when we are ready for review.

pdp7, 2020-01-13:
For those without KiCad:

Gerbers:

https://github.com/oshwabadge2020/Badge-PCB/tree/master/gerbers

Schematic PDF:

https://github.com/oshwabadge2020/Badge-PCB/tree/master/schematic.pdf

OSH Park renders:

Here is PDF of layer preview from OSH Park:

https://hackaday.io/project/168483-open-hardware-summit-2020-badge/files

ZIP of the layer renders:

https://cdn.hackaday.io/files/1684837208376672/OSH%20Park%20_%20Verify%20your%20design.zip

OSH Park shared project has hi res top and bottom render:

https://oshpark.com/shared_projects/DU7vUFIe

closed 2020-01-14 by pdp7