oshwabadge2020/Badge-PCB #15: review layout before fab

pdp7, 2020-01-13:
Thanks so much to @mwelling  for finishing routing the badge:
https://github.com/oshwabadge2020/Badge-PCB/pull/14

Michael used 2 layers for the initial prototypes so we can fab them faster.  There are only 8 weeks left until the Open Hardware Summit on March 13th.

For those without KiCad:
* Gerbers: https://github.com/oshwabadge2020/Badge-PCB/tree/master/gerbers
* Schematic PDF:https://github.com/oshwabadge2020/Badge-PCB/tree/master/schematic.pdf

OSH Park renders:
* Here is PDF of layer preview from OSH Park: https://hackaday.io/project/168483-open-hardware-summit-2020-badge/files
* ZIP of the layer renders: https://cdn.hackaday.io/files/1684837208376672/OSH%20Park%20_%20Verify%20your%20design.zip
* OSH Park shared project has hi res top and bottom render: https://oshpark.com/shared_projects/DU7vUFIe

Related Hackaday.io post:
https://hackaday.io/project/168483-open-hardware-summit-2020-badge/log/172923-help-review-layout-before-fab

pdp7, 2020-01-13:
@Junes-PhD @Anool @acamilo @morganallen

Anool, 2020-01-13:
1. MIC1 hole needs to be 1mm per the [datasheet](https://github.com/oshwabadge2020/Badge-PCB/blob/master/datasheets/mic_SPH0645LM4H-B.pdf)

2. Some of the legend text is getting subtracted by Via's. You might want to move them around a bit for better visibility.

3. You might want to add some descriptive text/link about the project on the PCB silk legend.

I'll get time to take a closer look tonight,

Anool, 2020-01-13:
4. U7 Gesture sensor has silk outline on both sides of the PCB

ShawnHymel, 2020-01-13:
I believe that Q1 needs to be a P-channel MOSFET (as opposed to an N-channel). The body diode on the FET needs to allow current from +BATT to VBUS when the USB is not plugged in. I usually use Jim's design here for reference when I'm doing similar power circuitry: https://cdn.sparkfun.com/assets/learn_tutorials/5/0/7/esp32-thing-schematic.pdf

ShawnHymel, 2020-01-13:
Are 2 10uF capacitors needed at the output of U2? The datasheet for the S-1318 (https://www.mouser.com/pdfdocs/S1318_E.pdf) claims that you only need a single 1.0 uF at the output, so 2x 10uF might be overkill (and 10uFs can be a little pricey).

ShawnHymel, 2020-01-13:
Can the S-1318 produce enough power? The datasheet says that it's limited to 100mA. In my experience, for the APDS-9960 to detect gestures, it needs to enable a "BOOST" IR LED that can suck up to 100 mA in short bursts.

ShawnHymel, 2020-01-13:
According to 6.1 in the BME680 datasheet (https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets_1/bst-bme680-ds001.pdf), you need to pull CSB up for I2C mode. Pulling it down activates SPI comms and disables I2C on the device.

ShawnHymel, 2020-01-13:
C15 may not be necessary if you've got bulk decoupling from the other 10uFs all over the board.

ShawnHymel, 2020-01-13:
What purpose do the D+ and D- test points serve? Will you actually be probing/sniffing USB traffic? I recommend not having test points for these lines, as they can create stubs on the differential pair for your data lines, which can cause reflections and mess up USB traffic.

schneider42, 2020-01-13:
Major: 
 - Display IM0,IM1,IM2 usage is different from the one in the Rev G schematic from Adafruit
 - CSB of BME680 needs to be tied high to select I2C
 - Partnumber of SPI flash does not match package (should be W25Q32JVZP I guess)
 
Minor:
 - Maybe put the output of D3 onto a pad
 - Check size of solder mask defined pads of 0402s when doing a VCC/GND pour

Questions:
 - LDO has abs. max of 120 mA. I guess it is enough. Just wanted to verify.
 - Is 3.0 V enough for the backlight? Datasheet says 3.0 V _typical_

I didn't check much in the layout yet. Might have some more time tomorrow.

mwelling, 2020-01-13:
It is appears that Q1 and Q3 should both be P-channel MOSFETs. Switched to AO3401A.
Q2 looks wired wrong; switching it to P-channel high side switch as well to keep the BOM simple.

Beefed up U2 by using AP2112K-3.3V which gives us 600mA and 3.3V 

CSB is now pulled up instead of down.

Switched the PN of U1 to W25Q32JVZPIQ.

Fixed the wiriing of IM0-IM2.

Looks like LCD_BL is not connected to a GPIO. Adding it.

Microphone pattern was not changes the AP diameter is as specified.

Soldermask defined pads should be okay.

We can tweak silkscreen and soldermask before final cut.

The test pads on the USB I assume are for use with a test fixture. Will check with Alex.

mwelling, 2020-01-13:
Updates are on my fork for verification:
https://github.com/mwelling/Badge-PCB

Please check my changes and make sure I didn't do something stupid.

mwelling, 2020-01-13:
The changes have been merged to this repo as well.

Junes-PhD, 2020-01-15:
*Probably minor notes*

RF layout recommendation for BMD-340 say to have a 4.4mm clear strip of
copper.  Wrapping the mic and copper pour around that antenna will reduce
range but not sure how much.  If the application is hand to phone it should
be fine but it's non-ideal for long mesh stuff and may need more broadcast
power.
BMD-340 Datasheet
<https://rigado.sharepoint.com/sites/Public/Shared%20Documents/Forms/AllItems.aspx?id=%2Fsites%2FPublic%2FShared%20Documents%2FWebsite%2FModules%2FBMD%2D300%2FBMD%2D340%2DData%2DSheet%2Dv2%2E1%2Epdf&parent=%2Fsites%2FPublic%2FShared%20Documents%2FWebsite%2FModules%2FBMD%2D300&p=true&originalPath=aHR0cHM6Ly9yaWdhZG8uc2hhcmVwb2ludC5jb20vOmI6L3MvUHVibGljL0VaSHBybnNaV21SRGl1eFByTUNMUFAwQmVCYjdjeFE0QmdSU3NIY3ZNdW9yR3c_cnRpbWU9VWhKOUYweVoxMGc>
pg. 21

3V3 maybe better without big flood.  I would at least prefer to switch to
thermal relief for more consistent temperature rise on pads.

USB crosses over SPI lines and doesn't have continuous ground plane

ESD protection  maybe a good idea. USB lines and Switch inputs.

mwelling, 2020-01-15:
Thanks @Junes-PhD! Will take a look.

FYI. I just flipped the source and drain on Q2 on my repo. This makes it a proper load switch.

We need to change the circuitry on the RBG LEDs as it's FET is backwards as well and has 5V going to the BMD inputs. Was thinking we could maybe use a tps22919 or similar for the load switch.

pdp7, 2020-01-15:
@mwelling I've merged in your fix for Q2 https://github.com/oshwabadge2020/Badge-PCB/commit/d8f25c52f77bb43815fccb44903c51dd587437b2  thank you

mwelling, 2020-01-16:
So we sent off the boards to fab for initial protos.

Thank you to everyone that reviewed the design.

schneider42, 2020-01-17:
Good Luck!

schneider42, 2020-01-17:
One more thing from my side:

The AP2112K has a rather high quiescent current of 55 uA when active (which it will be unless the battery is empty). That will limit how long the RTC is be able to run after the software has decided that it is time to go into deep sleep until the next charge. Loosing the current time shortly after the badge shuts down is annoying.

Maybe the TCR2LF30,LM(CT is an option (200 mA)? Of course this can also be evaluated on the prototype itself and changed in the final layout. If the APDS-9960 low pass stays like that I'd be surprised if it goes over about > 70 mA.

Not sure if this is a concern for you. If yes I'm happy to open a new issue.

mwelling, 2020-01-17:
We can evaluate this on the protos.