oshwabadge2020/Badge-PCB #27: 0603 capacitor specified for a 0402 pattern

mwelling, 2020-01-28:
https://github.com/oshwabadge2020/Badge-PCB/blob/master/badge.csv#L4

For the 10uF capacitors we have an 0402 footprint and an 0603 component specified in the BOM.

pdp7, 2020-02-10:
I am emailing kevin at digikey about this

mwelling, 2020-02-11:
We need to change the BOM inside the project to reflect an 0402 component.

pdp7, 2020-02-11:
@nseidle wrote:
> From: Nathan Seidle 
> Date: Tue, Jan 28, 2020 at 8:50 PM
> Subject: Re: Need an assembly sheet
> 
> FYI from my team:
> "Currently working on this, and it appears that either the wrong part was sourced or the package that was used for the layout was wrong. Either way, C15, C3, C4, C5, C6, and C7 are calling for a 10uF (0402) capacitor, but the supplier PN  311-1781-1-ND is a 0603 part (which is what was given to me)."
> I've asked my group to sub a 4.7uF 0402 to keep the machines going.

pdp7, 2020-02-11:
@mwelling What do you think of this one?

311-1729-1-ND
CAP CER 10UF 6.3V X5R 0402 +/- 20%
https://www.digikey.com/product-detail/en/yageo/CC0402MRX5R5BB106/311-1729-1-ND/5195631

pdp7, 2020-02-11:
> Subject: OHSummit: change 10UF cap from 0603 to 0402
> To: Kevin Walseth
> 
> Hi Kevin,
> 
> Sorry for the trouble, but we got this wrong in the BoM.
> 
> Qty 6 of 10 uF (C15 C3 C4 C5 C6 C7) should be 0402 but instead we
specified 0603:
> 
> 311-1781-1-ND
> CAP CER 10UF 6.3V X5R 0603 +/- 10%
https://www.digikey.com/product-detail/en/yageo/CC0603KRX5R5BB106/311-1781-1-ND/5195683
> 
> Could you ship 0402 version instead to SparkFun?
> 
> Maybe this one which is 0402?
> 
> 311-1729-1-ND
> CAP CER 10UF 6.3V X5R 0402 +/- 20%
https://www.digikey.com/product-detail/en/yageo/CC0402MRX5R5BB106/311-1729-1-ND/5195631
> 
> 
> Thanks,
> Drew

mwelling, 2020-02-12:
Looks good to me.

commit e39bc3d9 (2020-02-13): Fix #27 and #41 on BoM
closed 2020-02-13 by pdp7