oshwabadge2020/Badge-PCB #30: Feedback from SparkFun assembly of R0 prototype

pdp7, 2020-01-30:
Feedback from @nseidle from visual inspection:
* Please camel case SparkFun in the 'sponsored by' silk
* The FP for the Winbond IC looks incorrect (too skinny for wide IC)
* I expect to see more solder on the underside of the microB connector (image below). We usually have solder paste go into the anchor holes. This FP looks very dry that may lead to microB connectors being wrenched off. I have plugged in a cable and it feels fine. Just looks off to me.
* (nit pick) If we can use the Qwiic logo on the board instead of 'qwiic' I would be grateful. SVG of Qwiic logo attached.
* The protos have white Qwiic connectors. This is an error by SparkFun. I think Digikey shipped us connectors so my team used them rather than black ones. We'll get black ones on there for the final assembly.
* There are mouse bites under the microB connector. These will be next to impossible to remove on final assemblies. I recommend moving the connector or talking to OSHPark about moving them (I don't think that's possible but maybe Drew knows more).

pdp7, 2020-01-30:
> There are mouse bites under the microB connector. These will be next to impossible to remove on final assemblies. I recommend moving the connector or talking to OSHPark about moving them (I don't think that's possible but maybe Drew knows more).

@laen tells me that we will be able to specify tab keepout area on the production run

pdp7, 2020-01-30:
> (nit pick) If we can use the Qwiic logo on the board instead of 'qwiic' I would be grateful. SVG of Qwiic logo attached.

I have uploaded the logo to images dir in this repo:
https://github.com/oshwabadge2020/Badge-PCB/tree/master/images

@acamilo @mwelling do you think there is space?

acamilo, 2020-02-02:
I'll fit in the QWIIC connector.
I could hand solder the USB connectors over a weekend if it proves to be a problem.

closed 2020-02-07 by acamilo