oshwabadge2020/Badge-PCB #33: Short between 3.3V and SCL on some boards

mwelling, 2020-02-03:
2 of the 4 prototype units I was sent from Sparkfun had a short from 3.3V to SCL.

After inspecting the PCB layout, Alex noticed that the two nets are nearest under U5.

[image]

You will see that they directly adjacent and the solder mask is improperly specified for the pads allowing for easier shorts.

I recommend updating U5's pattern and possibly doing thermals on the power pour to avoid manufacturing issues. Couldn't hurt to check the paste apertures as well.

pdp7, 2020-02-05:
Is this fixed with https://github.com/oshwabadge2020/Badge-PCB/commit/0f4dcff1dcd95b07f539cf87d5421f1a353eb1b8 from @acamilo ?

pdp7, 2020-02-07:
@mwelling @acamilo Should I order new PCBs?

pdp7, 2020-02-07:
@nseidle Would you be able to look at the PCBs you received an check if there is short between TP5 to TP7, and TP6 to TP7?

We think the footprint on U5 was wrong.  Both the boards I received have I2C bus shorted.

mwelling, 2020-02-07:
We can leave this open until we find the source of the shorts.

siddacious, 2020-02-15:
probably not useful but I can confirm that VCC/3V and SCL are bridged on my board. I'll pull U5 and order a breakout from work for testing/dev until I can get an updated board

pdp7, 2020-02-15:
hopefully  this is resolved with the updated footprint for U5 on Rev1.

I just removed U5 from Rev0 so I could use i2c.

siddacious, 2020-02-15:
@pdp7 @mwelling here's a picture of my U5 pads after removal:
[image]
I'm near certain that this will be addressed by @acamilo 's fix

pdp7, 2020-02-15:
@acamilo did U5 work on your Rev1?

mwelling, 2020-02-15:
Yes it worked, he did the i2c scan yesterday and all of the devices appeared.