oshwabadge2020/Badge-PCB #46: BoM parts availabiity

Cipher2600, 2021-04-26:
I was attempting to gather the parts from DigiKey to create the badge as a project. Unfortunately, some of the parts are no longer available and/or have obscene long lead times. Drew recommended that I show which parts are no longer available and see if you have a recommended alternative. My knowledge on determining substitutes is limited to matching voltage, current, and form factor. If there is an important item that is not one of those, I will mess it up. Please see the attached spreadsheet with the listing of which parts are no longer available.

Any help would be most appreciated.
[OSN 2020 Badge BoM.xlsx](https://github.com/oshwabadge2020/Badge-PCB/files/6374222/OSN.2020.Badge.BoM.xlsx)

pdp7, 2021-04-26:
@acamilo @mwelling any thoughts about substitutions? thanks!

mwelling, 2021-04-28:
Sorry there is not much you can do about the BLE module, IMU, sensors. They are just not stocked right now.

Cipher2600, 2021-04-28:
That is what I thought. With the shortage of semiconductors worldwide, it hits everywhere. If possible, may I research possible alternatives and post them here to see if they would work? Like I said, I feel I am not fully knowledgeable to make that determination.

pdp7, 2021-04-29:
@Cipher2600 if you just wanted to develop with, then I should probably be able to send you one as I believe I have two.  Email me drew at oshpark.com and I will check _(moved from europe to USA in the mean time so I need to look in some boxes :)_.  

If you were interested in the process of DIY assembling one yourself, then I think the supply chain issues will make that not practical.

Cipher2600, 2021-05-21:
Drew,
Just curious if you were able to find one. I ordered the boards and they came in yesterday. I will probably get a stencil for some of the parts. 

If you are interested, I will keep you informed if I am successful in recreating the badge/watch.

Very Respectfully,
Anthony Monge

***Intercepted by the NSA. Read by the FBI.***